inivation/caer #161: modules/cameracalibration/CMakeFiles/cameracalibration.dir/calibration.cpp.o' failed

raghad1234, 2018-09-20:
Hi, 
I am trying to install the cAER library so I cloned it and followed the steps in the readme file. I got the following error when using $make command  

user@ubuntu:~/caer$ make
[ 19%] Built target caersdk
[ 19%] Built target caer-bin
[ 19%] Building CXX object modules/cameracalibration/CMakeFiles/cameracalibration.dir/calibration.cpp.o
/home/user/caer/modules/cameracalibration/calibration.cpp: In constructor ‘Calibration::Calibration(CameraCalibrationSettings)’:
/home/user/caer/modules/cameracalibration/calibration.cpp:3:60: warning: declaration of ‘settings’ shadows a member of ‘Calibration’ [-Wshadow]
 Calibration::Calibration(CameraCalibrationSettings settings) {
                                                            ^
In file included from /usr/include/strings.h:23:0,
                 from /usr/include/string.h:431,
                 from /usr/include/c++/7/cstring:42,
                 from /usr/include/libcaer/libcaer.h:22,
                 from /usr/include/libcaer/events/common.h:14,
                 from /usr/include/libcaer/events/frame.h:19,
                 from /home/user/caer/modules/cameracalibration/calibration.hpp:10,
                 from /home/user/caer/modules/cameracalibration/calibration.cpp:1:
/home/user/caer/modules/cameracalibration/calibration.hpp:34:39: note: shadowed declaration is here
  CameraCalibrationSettings settings = NULL;
                                       ^
/home/user/caer/modules/cameracalibration/calibration.cpp: In member function ‘void Calibration::updateSettings()’:
/home/user/caer/modules/cameracalibration/calibration.cpp:18:32: warning: comparing floating point with == or != is unsafe [-Wfloat-equal]
   if (settings->aspectRatio == 0.0f) {
                                ^~~~
/home/user/caer/modules/cameracalibration/calibration.cpp: In member function ‘double Calibration::computeReprojectionErrors(const std::vector<std::vector<cv::Point3_<float> > >&, const std::vector<std::vector<cv::Point_<float> > >&, const std::vector<cv::Mat>&, const std::vector<cv::Mat>&, const cv::Mat&, const cv::Mat&, std::vector<float>&, bool)’:
/home/user/caer/modules/cameracalibration/calibration.cpp:116:92: warning: declaration of ‘distCoeffs’ shadows a member of ‘Calibration’ [-Wshadow]
  const Mat &cameraMatrix, const Mat &distCoeffs, vector<float> &perViewErrors, bool fisheye) {
                                                                                            ^
In file included from /home/user/caer/modules/cameracalibration/calibration.cpp:1:0:
/home/user/caer/modules/cameracalibration/calibration.hpp:40:6: note: shadowed declaration is here
  Mat distCoeffs;
      ^~~~~~~~~~
/home/user/caer/modules/cameracalibration/calibration.cpp:116:92: warning: declaration of ‘cameraMatrix’ shadows a member of ‘Calibration’ [-Wshadow]
  const Mat &cameraMatrix, const Mat &distCoeffs, vector<float> &perViewErrors, bool fisheye) {
                                                                                            ^
In file included from /home/user/caer/modules/cameracalibration/calibration.cpp:1:0:
/home/user/caer/modules/cameracalibration/calibration.hpp:39:6: note: shadowed declaration is here
  Mat cameraMatrix;
      ^~~~~~~~~~~~
/home/user/caer/modules/cameracalibration/calibration.cpp:116:92: warning: declaration of ‘imagePoints’ shadows a member of ‘Calibration’ [-Wshadow]
  const Mat &cameraMatrix, const Mat &distCoeffs, vector<float> &perViewErrors, bool fisheye) {
                                                                                            ^
In file included from /home/user/caer/modules/cameracalibration/calibration.cpp:1:0:
/home/user/caer/modules/cameracalibration/calibration.hpp:38:26: note: shadowed declaration is here
  vector<vector<Point2f>> imagePoints;
                          ^~~~~~~~~~~
/home/user/caer/modules/cameracalibration/calibration.cpp:135:52: warning: conversion to ‘double’ from ‘size_t {aka long unsigned int}’ may alter its value [-Wconversion]
   perViewErrors[i] = (float) std::sqrt(err * err / n);
                                                    ^
/home/user/caer/modules/cameracalibration/calibration.cpp:140:31: warning: conversion to ‘double’ from ‘size_t {aka long unsigned int}’ may alter its value [-Wconversion]
  return (std::sqrt(totalErr / totalPoints));
                               ^~~~~~~~~~~
/home/user/caer/modules/cameracalibration/calibration.cpp: In member function ‘void Calibration::calcBoardCornerPositions(cv::Size, float, std::vector<cv::Point3_<float> >&, CameraCalibrationPattern)’:
/home/user/caer/modules/cameracalibration/calibration.cpp:144:103: warning: declaration of ‘boardSize’ shadows a member of ‘Calibration’ [-Wshadow]
  Size boardSize, float squareSize, vector<Point3f> &corners, enum CameraCalibrationPattern patternType) {
                                                                                                       ^
In file included from /home/user/caer/modules/cameracalibration/calibration.cpp:1:0:
/home/user/caer/modules/cameracalibration/calibration.hpp:36:7: note: shadowed declaration is here
  Size boardSize;
       ^~~~~~~~~
/home/user/caer/modules/cameracalibration/calibration.cpp:152:36: warning: conversion to ‘float’ from ‘int’ may alter its value [-Wconversion]
      corners.push_back(Point3f(x * squareSize, y * squareSize, 0));
                                    ^~~~~~~~~~
/home/user/caer/modules/cameracalibration/calibration.cpp:152:52: warning: conversion to ‘float’ from ‘int’ may alter its value [-Wconversion]
      corners.push_back(Point3f(x * squareSize, y * squareSize, 0));
                                                    ^~~~~~~~~~
/home/user/caer/modules/cameracalibration/calibration.cpp:160:39: warning: conversion to ‘float’ from ‘int’ may alter its value [-Wconversion]
      corners.push_back(Point3f((2 * x + y % 2) * squareSize, y * squareSize, 0));
[... 67 lines cut ...]
    undistortEventInputMap.push_back(Point2f(x + 0.5, y + 0.5));
                                                      ~~^~~~~
/home/user/caer/modules/cameracalibration/calibration.cpp: In member function ‘void Calibration::undistortEvent(caerPolarityEvent, caerPolarityEventPacket)’:
/home/user/caer/modules/cameracalibration/calibration.cpp:408:50: warning: conversion to ‘uint16_t {aka short unsigned int}’ from ‘int’ may alter its value [-Wconversion]
   caerPolarityEventSetX(polarity, eventUndistort.x);
                                   ~~~~~~~~~~~~~~~^
/home/user/caer/modules/cameracalibration/calibration.cpp:409:50: warning: conversion to ‘uint16_t {aka short unsigned int}’ from ‘int’ may alter its value [-Wconversion]
   caerPolarityEventSetY(polarity, eventUndistort.y);
                                   ~~~~~~~~~~~~~~~^
modules/cameracalibration/CMakeFiles/cameracalibration.dir/build.make:86: recipe for target 'modules/cameracalibration/CMakeFiles/cameracalibration.dir/calibration.cpp.o' failed
make[2]: *** [modules/cameracalibration/CMakeFiles/cameracalibration.dir/calibration.cpp.o] Error 1
CMakeFiles/Makefile2:236: recipe for target 'modules/cameracalibration/CMakeFiles/cameracalibration.dir/all' failed
make[1]: *** [modules/cameracalibration/CMakeFiles/cameracalibration.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

Is there a problem in the library itself or it is a compatibility issue? I am using the following versions 
libcaer version: 2.5.0-1~bionic
boost 1.65.1.0
cmake v3.10.2
gcc (Ubuntu 7.3.0-16ubuntu3) 7.3.0

Appreciate your help.
Regards,
Raghad

llongi, 2018-09-21:
The cause of the compilation failure seems to be:

`calibration.cpp:269:3: error: ‘cvWriteComment’ was not declared in this scope`

That's an OpenCV function. What version of OpenCV do you have installed on your system?

raghad1234, 2018-09-22:
I am using 4.0.0 openCV version

llongi, 2018-09-23:
Ah I see. OpenCV 4.0 only just became available as an alpha release, so we don't support it yet.
We only tested and support OpenCV from 3.1.0 to 3.4.3.
Can you try adding:

`#include <opencv2/core/core_c.h>`

to the top of the 'calibration.cpp' file and see if that helps?

raghad1234, 2018-09-23:
It worked, but I have lots of warning, is this ok? or it is better to use the previous tested OpenCV versions? 
part of the warnings I am getting is shown below 
/home/user/caer/utils/ext/linenoise-ng/linenoise.cpp:1877:40: warning: conversion to ‘char’ from ‘unsigned int’ may alter its value [-Wconversion]
           friendlyTextBuf[1] = keyCopy + 0x40;
                                ~~~~~~~~^~~~~~
/home/user/caer/utils/ext/linenoise-ng/linenoise.cpp: In function ‘void linenoisePrintKeyCodes()’:
/home/user/caer/utils/ext/linenoise-ng/linenoise.cpp:3413:17: warning: conversion to ‘int’ from ‘ssize_t {aka long int}’ may alter its value [-Wconversion]
     nread = read(STDIN_FILENO, &c, 1);
             ~~~~^~~~~~~~~~~~~~~~~~~~~
/home/user/caer/utils/ext/linenoise-ng/linenoise.cpp: In function ‘int linenoiseHistoryAdd(const char*)’:
/home/user/caer/utils/ext/linenoise-ng/linenoise.cpp:3451:1: warning: assuming signed overflow does not occur when changing X +- C1 cmp C2 to X cmp C2 -+ C1 [-Wstrict-overflow]
 }
 ^
/home/user/caer/utils/ext/linenoise-ng/linenoise.cpp: In function ‘void dynamicRefresh(PromptBase&, char32_t*, int, int)’:
/home/user/caer/utils/ext/linenoise-ng/linenoise.cpp:1184:24: warning: assuming signed overflow does not occur when simplifying conditional [-Wstrict-overflow]
   if (xEndOfInput == 0 && yEndOfInput > 0)
       ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~
/home/user/caer/utils/ext/linenoise-ng/linenoise.cpp: In member function ‘void InputBuffer::refreshLine(PromptBase&)’:
/home/user/caer/utils/ext/linenoise-ng/linenoise.cpp:1337:24: warning: assuming signed overflow does not occur when simplifying conditional [-Wstrict-overflow]
   if (xEndOfInput == 0 && yEndOfInput > 0)
       ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~
[ 90%] Building CXX object utils/caerctl/CMakeFiles/caer-ctl.dir/caerctl.cpp.o
/home/user/caer/utils/caerctl/caerctl.cpp: In function ‘void handleInputLine(const char*, size_t)’:
/home/user/caer/utils/caerctl/caerctl.cpp:276:28: warning: ISO C++ forbids variable length array ‘bufCopy’ [-Wvla]
  char bufCopy[bufLength + 1];
                            ^
/home/user/caer/utils/caerctl/caerctl.cpp: In function ‘void handleCommandCompletion(const char*, linenoiseCompletions*)’:
/home/user/caer/utils/caerctl/caerctl.cpp:581:28: warning: ISO C++ forbids variable length array ‘bufCopy’ [-Wvla]
  char bufCopy[bufLength + 1];
                            ^
/home/user/caer/utils/caerctl/caerctl.cpp:574:13: warning: stack protector not protecting local variables: variable length buffer [-Wstack-protector]
 static void handleCommandCompletion(const char *buf, linenoiseCompletions *autoComplete) {
             ^~~~~~~~~~~~~~~~~~~~~~~
/home/user/caer/utils/caerctl/caerctl.cpp: In function ‘void handleInputLine(const char*, size_t)’:
/home/user/caer/utils/caerctl/caerctl.cpp:271:13: warning: stack protector not protecting local variables: variable length buffer [-Wstack-protector]
 static void handleInputLine(const char *buf, size_t bufLength) {
             ^~~~~~~~~~~~~~~
[ 92%] Linking CXX executable caer-ctl
Warning: Unused direct dependencies:
        /usr/lib/x86_64-linux-gnu/libcaer.so.2
        /usr/lib/x86_64-linux-gnu/libboost_iostreams.so.1.65.1
        /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.65.1
        /lib/x86_64-linux-gnu/libdl.so.2
        /lib/x86_64-linux-gnu/libm.so.6
[ 92%] Built target caer-ctl
Scanning dependencies of target tcpststat
[ 93%] Building C object utils/tcpststat/CMakeFiles/tcpststat.dir/tcpststat.c.o
In file included from /usr/include/x86_64-linux-gnu/sys/types.h:194:0,
                 from /usr/include/stdlib.h:394,
                 from /home/user/caer/ext/net_rw.h:6,
                 from /home/user/caer/utils/tcpststat/tcpststat.c:1:
/home/user/caer/modules/inout/inout_common.h: In function ‘caerGenericEventSetTimestamp’:
/home/user/caer/modules/inout/inout_common.h:10:13: warning: conversion to ‘__uint32_t {aka unsigned int}’ from ‘int32_t {aka int}’ may change the sign of the result [-Wsign-conversion]
   = htole32(timestamp);
             ^
In file included from /home/user/caer/utils/tcpststat/tcpststat.c:2:0:
/home/user/caer/modules/inout/inout_common.h:10:5: warning: conversion to ‘int32_t {aka int}’ from ‘__uint32_t {aka unsigned int}’ may change the sign of the result [-Wsign-conversion]
   = htole32(timestamp);
     ^~~~~~~
/home/user/caer/utils/tcpststat/tcpststat.c: In function ‘main’:
/home/user/caer/utils/tcpststat/tcpststat.c:170:23: warning: initialization discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
    void *firstEvent = caerGenericEventGetEvent(header, 0);
                       ^~~~~~~~~~~~~~~~~~~~~~~~
/home/user/caer/utils/tcpststat/tcpststat.c:171:23: warning: initialization discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
    void *lastEvent  = caerGenericEventGetEvent(header, eventValid - 1);
                       ^~~~~~~~~~~~~~~~~~~~~~~~
[ 94%] Linking C executable tcpststat
[ 94%] Built target tcpststat
Scanning dependencies of target udpststat
[ 96%] Building C object utils/udpststat/CMakeFiles/udpststat.dir/udpststat.c.o
[... 29 lines cut ...]
[ 97%] Linking C executable udpststat
[ 97%] Built target udpststat
Scanning dependencies of target unixststat
[ 98%] Building C object utils/unixststat/CMakeFiles/unixststat.dir/unixststat.c.o
In file included from /usr/include/x86_64-linux-gnu/sys/types.h:194:0,
                 from /usr/include/stdlib.h:394,
[... cut ...]
/home/user/caer/utils/unixststat/unixststat.c:149:23: warning: initialization discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
    void *firstEvent = caerGenericEventGetEvent(header, 0);
                       ^~~~~~~~~~~~~~~~~~~~~~~~
/home/user/caer/utils/unixststat/unixststat.c:150:23: warning: initialization discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
    void *lastEvent  = caerGenericEventGetEvent(header, eventValid - 1);
                       ^~~~~~~~~~~~~~~~~~~~~~~~
[100%] Linking C executable unixststat
[100%] Built target unixststat

llongi, 2018-09-24:
Glad to hear it works now, I've committed the fix in commit 58f8b3f5e571bb13b60e60f05b2c64f86d462086.
The warnings are fine, they're mostly due to external code we use.
Closing this.

closed 2018-09-24 by llongi